arduino/Arduino #11104: readme-edits

Maclain272, 2020-12-26:
> Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2020-12-26:
@Maclain272 please stop doing this. You're being very rude.

closed 2020-12-26 by per1234

per1234, 2020-12-27:
@Maclain272 since you have continued with your disrespectful behavior, I'll ask again: please stop creating meaningless issues, PRs, PR reviews, comments, and wiki pages in the repositories of Arduino and other open source organizations. You are creating a large maintenance burden for the maintainers of these repositories, and only making yourself look bad, so what is the point? If you continue, I'll be forced to report you to GitHub and block you from all the Arduino repositories.

Maclain272, 2020-12-27:
Because i am trying self balancing robot in Arduino nano but I installed the library already but it till need support to read it like lmotorcontroller isn't!!!!
 I am sorry tell me what to do?

Maclain272, 2020-12-27:
Please i need a help !

per1234, 2020-12-27:
GitHub is a place for people to collaborate and share. It is not a place to ask for assistance with your project.

The place to ask for assistance with Arduino projects is the Arduino Forum:
http://forum.arduino.cc/index.php?topic=148850
we can definitely help you out over there. But be warned that we won't put up with disrespectful behavior on the Arduino Forum either. If you behave there the way you have been here on GitHub, I can promise you'll be banned immediately.

If you'll straighten yourself out and realize that you're interacting with real people, not just unfeeling pixels on a screen, you'll be welcome as part of the Arduino community.