arduino/Arduino #11111: [Library Manager] Please add the following libraries to the Arduino library manager

lakshanthad, 2020-12-29:
Hi, 
I would appreciate it if you could add the following libraries to the Arduino library manager.

https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi
https://github.com/Seeed-Studio/Seeed_Arduino_rpcUnified
https://github.com/Seeed-Studio/Seeed_Arduino_FS
https://github.com/Seeed-Studio/Seeed_Arduino_SFUD
https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR
https://github.com/Seeed-Studio/Seeed_Arduino_IR

per1234, 2020-12-29:
Hi @lakshanthad. Thanks for your contributions to the Arduino Library Manager index!

Libraries added to the Library Manager index are required to define a URL where users can get more information about the library in the `url` field of library.properties. The "Seeed Arduino FS" and "Seeed Arduino IR" libraries have blank `url` fields, which is not allowed:
- https://github.com/Seeed-Studio/Seeed_Arduino_FS/blob/v2.0.1/library.properties#L8
- https://github.com/Seeed-Studio/Seeed_Arduino_IR/blob/v1.0.1/library.properties#L8

Please update the `url` values in these libraries (and fill in the rest of the Seeed Arduino IR library's library.properties while you're at it), then either update the `version` values and create new tags/releases or else redo the existing releases.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

---
Regarding https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR , you'll be happy to learn it was already added by https://github.com/arduino/Arduino/issues/10914

lakshanthad, 2020-12-29:
Hi, 
Thank you for pointing out the Seeed_Arduino_LIS3DHTR.

I have made the necessary changes to Seeed_Arduino_FS and Seeed_Arduino_IR libraries according to the above-mentioned requirements. Please check. 

Thank you

lakshanthad, 2021-01-04:
@per1234 Can I know when the libraries can be added? 

Also this library:
https://github.com/arduino/Arduino/issues/11113

Best Regards,
Lakshantha

per1234, 2021-01-04:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

lakshanthad, 2021-01-04:
Thank you for the clarification

cmaglie, 2021-01-09:
Done.

closed 2021-01-09 by cmaglie

lakshanthad, 2021-01-09:
Thank you. Appreciate it!