arduino/Arduino #11122: Serial port mointor randomly freezes up

imranmoezkhan, 2021-01-01:
I'm using an MKR1010 and am trying to run the SerialEvent example: https://www.arduino.cc/en/Tutorial/BuiltInExamples/SerialEvent

I recently updated to the 1.8.13 release a few days ago, and I'm trying to send characters using the serial port monitor as required in the serial even example. The serial port monitor it freezes up after a random number of characters - sometimes 1 sometimes 3. I haven't received any output over the serial port monitor as the sketch should provide.

Also - this might be a separate issue - the library manager is throwing Java concurrent execution errors when I try to open it.

imranmoezkhan, 2021-01-02:
I rolled back to 1.8.9 (which was previously working reasonably stable for me), and now its gone completely haywire. Every time I try to upload a sketch Java throws a serialPortException - setEventsMask error. Pressing the reset button even removing then reconnecting the board has no effect, and multiple COM ports seem to pop up:
[image]

I have to restart Arduino studio to be able to upload a sketch. This is getting extremely frustrating...

imranmoezkhan, 2021-01-02:
It appears I am not the only one having this issue: https://forum.arduino.cc/index.php?topic=609249.0
For me, this serial port disappearing has happened after moving to the 1.8.13 ide release. Didn't see any Java serialport exceptions in the earlier 1.8.9 install before (had been using that for 4+ months) - don't know if its related.
Any support/workaround available for this, or just need to accept that the $700 testbed I've been working on for the past 6 months is the wrong way to go?

per1234, 2021-01-02:
>can this be officially confirmed?

Yes.

`serialEvent()` may be deprecated in the near future:
https://github.com/arduino/ArduinoCore-avr/issues/206#issuecomment-532133626

But that's no loss because It's absolutely trivial to implement this functionality in the sketch:
```c++
if (Serial.available()) {
  serialEvent();
}
```

> I don't see it in the function's documentation:

It's right here:
https://www.arduino.cc/reference/en/language/functions/communication/serial/serialevent/#_notes_and_warnings
>serialEvent() and serialEvent1() don’t work on the Arduino SAMD Boards

Work is in progress to even more clearly document the support for `serialEvent`:
https://github.com/arduino/ArduinoCore-avr/issues/206#issuecomment-727796961

>multiple COM ports seem to pop up:

See: https://github.com/arduino/Arduino/issues/8851

>Any support/workaround available for this

The workaround is to go back to using Arduino IDE 1.8.13, which has https://github.com/arduino/Arduino/issues/8851 fixed. Then update your sketch so it doesn't rely on the unsupported `serialEvent()` function. After that, I think you'll find everything works as expected.

---
Even though it won't occur with a correctly functioning sketch, the bug you reported is legitimate, since the Serial Monitor should never hang. A minimal demonstration of the problem is to upload any sketch that doesn't clear the serial RX buffer to a native USB board. For example:
```c++
void setup() {}
void loop() {}
```
Then open Serial Monitor and send input a few times. It will now freeze and you'll need to disconnect the board from the computer to unfreeze it.

Let's confine this issue exclusively to further discussion of this original bug, since it seems to be going off on some unrelated tangents now that will make it difficult for the developers to investigate.

imranmoezkhan, 2021-01-03:
Rolled forward to 1.8.13 again and still had this issue. Then wrote a "completely clean" sketch, and it seems to be working alright. One of the things I am testing right now with my sketch for my testbed is serial handling for quite a complicated command flow from my pc to control the mkr1010 - so its possible in my sketch I am not reading/clearing the buffers correctly and this is causing the monitor to freeze. Unfortunately, that means any mistake I make needs me to restart the ide. Not ideal, but its better than 1.8.9 where I needed to restart my pc. Thanks for your help. Hopefully the serial monitor bug gets fixed in a release.
--------------------
Just have another question though:
From the serialEvent() documentation: "Called when data is available. Use Serial.read() to capture this data."
Sorry, the arduinos are one of 4 different pieces of hardware I'm coding for my testbed in 4 different languages, so I may have understood the doc wrong - to me the language from the description in the doc sounds like a callback, and wouldn't be implementable with what you have suggested using Serial.available() (I believe Serial.available() is not called when serial data is available, but needs to be checked instead?).

per1234, 2021-01-03:
You can see the implementation of `serialEvent()` here:
- https://github.com/arduino/ArduinoCore-avr/blob/1.8.3/cores/arduino/main.cpp#L47
- https://github.com/arduino/ArduinoCore-avr/blob/1.8.3/cores/arduino/HardwareSerial.cpp#L64-L67

For further discussion of the code for your project, the Arduino Forum is more appropriate. I'll be happy to participate in that, and some of the forum members are very knowledgeable in this subject:
https://forum.arduino.cc/

imranmoezkhan, 2021-01-03:
Thanks for the link to the implementation - looks like serialEvent is only called at the end of the loop, which unfortunately is quite different than what the doc says.
Is ArduinoCore managed separately from arduino ide?