arduino/Arduino #11128: Arduino IDE 1.8.13 Won't Launch

dbjurstr, 2021-01-03:
I've been using Arduino IDE for years in Windows 7, 8, and 10. Now suddenly it won't launch in Windows 10. I've uninstalled it several times and re-installed it. It looks like it's going to run by the initial launch screen popping up but then it goes away and that's it, nothing.

I've run the cmd line arduino_debug.exe file and attached it. Good luck it's all mumbo jumbo to me.

[Arduino_Debug_Log.txt](https://github.com/arduino/Arduino/files/5762443/Arduino_Debug_Log.txt)

This has become a real mystery to me. HELP

per1234, 2021-01-04:
The relevant part of the debug output:
```
java.io.IOException: The cloud file provider is not running
```
@dbjurstr, this issue will give you the information you need to fix the problem with your computer: https://github.com/arduino/Arduino/issues/8283

closed 2021-01-04 by per1234

dbjurstr, 2021-01-04:
Okay, I think I got Arduino working. I tried to follow the steps in your attached link. Unfortunately It didn’t help. Which doesn’t surprise me. I’m can be pretty dangerous with a little knowledge. 

 

However, like I said I got it working. Just short of doing a complete drive restoration from a backup, I thought I’d go and uninstall the last Windows Update. Then I used jv16 Power Tools and did a search of installed software. I found 7 items that were residual pieces of software that I thought I had removed. One of them was Arduino. I removed it and rebooted.  I reloaded Arduino and it came up and runs as advertised. 

 

Thanks for the link and rapid response. I should have asked a few day ago. Ha, oh well. 

 

Thanks, Dennis

 

Closed #11128 <https://github.com/arduino/Arduino/issues/11128> .