arduino/Arduino #11131: Board Manager Update results in java.lang.NullPointerException

bdlabitt, 2021-01-04:
IDE 1.8.13, Platform linux64, Ubuntu 20.04.1
Have not be able to update boards using board manager.  This error is the same as reported #10335 (and closed) last year.  The apparent solution does not appear to be robust, since it has reoccured.  I cannot update the Arduino SAMD board support package.  Currently at 1.8.9.  Clicking update results in the error.

`java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionInstaller.remove(ContributionInstaller.java:276)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:170)
	... 1 more
`
Is the solution still to delete .arduino15/cache/cache.json and ./arduino15/package_index.json and restart the Arduino IDE, as stated by facchinm on June 10, 2020?

bdlabitt, 2021-01-04:
No, to answer my question, this fix does not work.  Renaming the above files, does not fix the error.

matthijskooijman, 2021-01-04:
I think this is a completely different problem. The `NullPointerException` error is just a generic indication that *some* variable is null when it shouldn't be, and looking at the backtrace, the cause is completely different.

Looking at the backtrace, the exception is caused here: https://github.com/arduino/Arduino/blob/5adf408270851e3739dab1b316dea62faaddd3c3/arduino-core/src/cc/arduino/contributions/packages/ContributionInstaller.java#L275-L276
Called from here: https://github.com/arduino/Arduino/blob/5adf408270851e3739dab1b316dea62faaddd3c3/app/src/cc/arduino/contributions/packages/ui/ContributionManagerUI.java#L170

So it seems it's either `destFolder` that is `null`, or `destFolder.getParentFile()` is `null` (I think [`getParentFile()`](https://docs.oracle.com/javase/7/docs/api/java/io/File.html#getParentFile()) can return null when `destFolder` is already the root path `/` or maybe an empty path).

I haven't dug further, so I have no advice on what could cause this or how to work around this, but hopefully the above helps others to figure out what's going on...