arduino/Arduino #11141: Mbed board packages for Nano 33 BLE are conflicting and out of date

petewarden, 2021-01-06:
I'm trying to track down an issue with TensorFlow Lite Micro that's occurring for users of the Arduino IDE, but doesn't show up with the CLI that we use for continuous integration testing. I've narrowed the cause of the disparity to there being two different board packages for the Nano 33 BLE, but it's unclear which one is the correct one to use. I'm filing this bug to try to answer that question, so we can unblock our CI system and ship a new version of the TFL Micro library.

In the Arduino IDE, typing "mbed" into the board manager search brings up two different options:
[image]

The second one is marked as deprecated, so I (and I assume most users) install the first one titled "Arduino nRF528x Boards (Mbed OS)". When that library has been installed, the title actually changes to "Arduino mbed-enabled boards":

[image]

From inspection of the downloaded files (for example platform.txt, etc), this is actually based on the repository here:
https://github.com/arduino/ArduinoCore-nRF528x-mbedos

Surprisingly, this repository's README indicates that it's deprecated in favor of another repo:
https://github.com/arduino/ArduinoCore-mbed

It also includes an out-dated version of the Mbed OS, which doesn't include a fix TFL Micro needs for a CMSIS bug (https://github.com/ARMmbed/mbed-os/issues/12568).

So the first big question is if the repo's README or the board manager index are correct - should the "nRF528x" board package be used or is it deprecated?

The second cause of confusion is that using the Arduino CLI with an FBQN of arduino:mbed:nano33ble specified appears to install the second package shown in the board manager screenshot, which seems to be based on https://github.com/arduino/ArduinoCore-mbed and includes a newer version of Mbed OS. This means when we're running our CI tests, we don't hit the Mbed/CMSIS bug, even though users of the IDE will. If the second board package is deprecated, I'd expect the first package to be chosen instead. This would cause our build to break, but at least it would reflect the typical user experience.

We'll go ahead and try to get a new release of the TFL Micro library out that includes a work-around patch for the CMSIS issue, but it would be very helpful if you could offer us some advice on what board package we should be using in the IDE, and what the expected behavior of the CLI is.

per1234, 2021-01-07:
Hi @petewarden.

Some background on the history of the Arduino Mbed OS boards packages:

The first Arduino board using Mbed OS was the Nano 33 BLE. The "Arduino nRF528x Boards (Mbed OS)" boards package (or `arduino:mbed` in machine readable form) was created for this board.

Some time later, the second Arduino board using Mbed OS was created: the Portenta H7. Some significant changes were required to add support for the Portenta to the existing boards platform. In order to provide support for the users of the Portenta while avoiding subjecting the Nano 33 BLE users to any possible breakage for that board that was introduced with the changes needed for the Portenta, a beta phase boards package was released: "Arduino mbed-enabled Boards" (or `arduino-beta:mbed` in machine readable form).

After some time, the  `arduino-beta:mbed` boards package was determined to be stable. Since this boards package was an advancement over the "Arduino nRF528x Boards (Mbed OS)" package, "Arduino nRF528x Boards (Mbed OS)" was deprecated. At the same time the machine readable identifier of the  `arduino-beta:mbed` boards package was changed from to the final  `arduino:mbed` and the  `arduino-beta:mbed` package was also deprecated. So this is why you see that both  "Arduino nRF528x Boards (Mbed OS)" and  `arduino-beta:mbed` are deprecated.

So you should be using `arduino:mbed@1.3.1`. There is a bug in the Arduino IDE's Boards Manager search feature that is causing some problems for people trying to find the Mbed OS package, which will be fixed by https://github.com/arduino/Arduino/pull/11096. However, I don't know where this 1.9.9 version is coming from, since there is no such version in [the primary Boards Manager package index](https://downloads.arduino.cc/packages/package_index.json) and I can't reproduce this result on my computer. Maybe you have a staging package index URL  in your preferences or a custom beta version of the package installed manually? You might try again without using the Boards Manager search feature. There aren't really so many packages, so it's not much extra effort to scroll through the full list to find "Arduino mbed-enabled Boards". If that doesn't help, try it again with the Arduino IDE in[ portable mode](https://www.arduino.cc/en/Guide/PortableIDE) to eliminate any possible interference from modifications.

>So the first big question is if the repo's README or the board manager index are correct

Both are correct.

> should the "nRF528x" board package be used or is it deprecated?

It's deprecated.

>The second cause of confusion is that using the Arduino CLI with an FBQN of arduino:mbed:nano33ble specified appears to install the second package shown in the board manager screenshot, which seems to be based on https://github.com/arduino/ArduinoCore-mbed and includes a newer version of Mbed OS.

That's the FQBN you should use and the correct behavior from Arduino CLI.

>at least it would reflect the typical user experience.

From what I've seen while providing support and maintaining the issue trackers since the transition of the Mbed OS boards packages, I don't think that is the typical user experience. Although some have had problems, I think most users have been able to install `arduino:mbed@1.3.1`.

petewarden, 2021-01-07:
> Maybe you have a staging package index URL in your preferences

Thanks so much for the suggestion, I believe that's it! The machine I'm using for testing the desktop IDE is the same one I had when I was testing an early version of the Nano 33 BLE, and I'd forgotten that one of the setup steps at that point was adding https://downloads.arduino.cc/packages/package_mbed_index.json to the preferences as a URL. I'd been deleting and reinstalling the application itself, but that still carried over in the preferences.

I've just confirmed that a search for "mbed" now only brings up the expected two packages, both named "Arduino mbed-enabled boards", with the second one deprecated. I'm able to install 1.3.1, and this does contain the Mbed fix I need.

Thanks again for your help, this will let us ship a new version of TFL Micro without requiring a patch. I'm closing this issue.

closed 2021-01-07 by petewarden