arduino/Arduino #11144: M1 Mac crashing when uploading code

phandrieu, 2021-01-07:
Hello, 
I reccently bought a M1 MacBook Pro, and tried a few Arduino IDE function, such as serial monitor, compiling code,... Everything worked fine, but yesterday I tried to upload code on an Arduino UNO, and after a few seconds, the whole system crashed, and rebooted. I tried to upload the same code on the same board using Windows, and it worked perfectly, so I am stuck with that...
Does anyone have encountered that kind of problem, or does anyone have a solution to this ?
Thanks :)

adeeb10abbas, 2021-01-15:
I think we need a new label for Apple Silicon issues @per1234. This seems similar to #10836 
This issue is most likely because of the lack of an apple silicon native IDE

mcmehrtens, 2023-04-10:
Has there been any progress towards a resolution of this bug? I'm trying to develop with Arduino IDE on my M1 MBP for an Adafruit Clue board, and I regularly get full computer crashes when uploading code to the board. I'm happy to provide log files if I'm told where to find them.

per1234, 2023-04-10:
Hi @mcmehrtens. If you haven't already, please download the Apple Silicon build of Arduino IDE 2.0.4 from the "**macOS Apple Silicon, 11: “Big Sur” or newer, 64 bits**" link on the Software page:

https://www.arduino.cc/en/software

If you still experience the crash using that version of Arduino IDE, you can submit a report in the Arduino IDE 2.x issue tracker:

https://github.com/arduino/arduino-ide/issues

I'll go ahead and close this issue as obsolete because the 1.x series of Arduino IDE tracked in this issue tracker has been replaced by the Arduino IDE 2.x series, which has a separate dedicated issue tracker.

closed 2023-04-10 by per1234

adeeb10abbas, 2023-04-10:
@mcmehrtens FWIW I have found platformIO + VSCode to be significantly better. ymmv