arduino/Arduino #11145: [Library Manager] Arduino Home Assistant integration (MQTT)

dawidchyrzynski, 2021-01-07:
Repository URL: https://github.com/dawidchyrzynski/arduino-home-assistant

per1234, 2021-01-07:
Hi @dawidchyrzynski. Thanks for your interest in contributing to the Arduino Library Manager index.

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties:
https://github.com/dawidchyrzynski/arduino-home-assistant/blob/1.0.0/library.properties#L1
```
name=arduino-home-assistant
```
to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

dawidchyrzynski, 2021-01-08:
Hi @per1234, thank you for your message. Feel free to review the 1.0.1 tag.

cmaglie, 2021-01-09:
Done.

closed 2021-01-09 by cmaglie