arduino/Arduino #11150: Arduino IDE 1.8.13 menu bar and icon options text missing in Archlinux LXQt

severnaside, 2021-01-10:
Arduino 1.8.13 (using your Software Download) was working ok until I did an Arch system upgrade. See screenshot below. The options are there, but not visible.  If I click on "Open", for example, and blindly follow down and click on something - a sketch will open.  

I did a system downgrade (rollback) using the Arch Archive to the previous upgrade date, and the IDE worked correctly again.  I don't know which package(s) caused the problem as there were 124 packages upgraded after a 3-week interval (remiss of me). So, I brought my system back up-to-date and tried something else:

In the IDE download folder there is a folder named 'java'. I renamed it to something different  and launched the IDE - everything working ok.  **This is my solution for now.**

Being curious, I also installed the Arch Arduino 1:1.8.13-2 package using pacman package manager .  I note that the 'arduino' folder /usr/share/arduino does not contain a folder 'java'. I prefer to use your Arduino Software, so I will stay with that.

I have another machine running Archlinux, it behaved the same way as above.

You may not consider this to be an arduino.cc issue, but I just wanted to make you aware. 

OS: Archlinux 64-bit
Desktop: LXQt
Java Environment: java-14-openjdk

[image]

lefeverd, 2021-01-11:
Same issue here on Fedora 5.9.16-200.fc33.x86_64
openjdk version "11.0.9.1" 2020-11-04
OpenJDK Runtime Environment 18.9 (build 11.0.9.1+11)

Installing the latest java with `sudo dnf install java-latest-openjdk` (and setting it as default with `sudo alternatives --config java`), in addition to renaming the java folder inside the arduino directory solves the issue for now.

facchinm, 2021-01-12:
As @lefeverd reported, it's likely to be a mismatch between the installed java version and the one used to build the IDE. The distribution from arduino.cc should be unaffected (uses the provided `java` folder as base jre installation) but there could be issues with other components (gtk bindings or similar).
Can any of you run the full IDE from arduino.cc (keeping the `java` folder) from the commandline and paste the console log here? Thanks!

severnaside, 2021-01-12:
@facchinm, see my console output:
`[chris@crt2 ~]$ arduino
Picked up JAVA_TOOL_OPTIONS:
Set log4j store directory /home/chris/.arduino15`

It is the same, with or without the java folder.
The "Picked up JAVA_TOOL_OPTIONS:" message is just what java (openjdk) writes to stderr, according to ArchWiki -Java.

If I wait approx. 1 minute, there is further output which also goes to the .arduino15 log, but this only seems to be application/url checks, see extract below:
`2021-01-12T12:25:03.229Z INFO c.a.c.p.ContributionInstaller:305 [Timer-0] Start download and signature check of=[http://drazzy.com/package_drazzy.com_index.json, https://downloads.arduino.cc/packages/package_index.json]
2021-01-12T12:25:03.251Z INFO c.a.u.n.FileDownloaderCache:92 [Timer-0] Cache folder /home/chris/.arduino15/cache
2021-01-12T12:25:03.305Z INFO c.a.u.n.FileDownloaderCache:112 [Timer-0] Number of file already in the cache 4
2021-01-12T12:25:03.347Z INFO c.a.u.n.FileDownloaderCache:149 [Timer-0] Get file cached is expire true, exist false, info FileCached{eTag='null', lastETag='null', remoteURL='http://drazzy.com/package_drazzy.com_index.json', localPath='/home/chris/.arduino15/cache/drazzy.com/package_drazzy.com_index.json', md5='null', createdAt='2021-01-12T12:25:03.32', cacheControl=null} 
2021-01-12T12:25:03.357Z INFO c.a.u.n.HttpConnectionManager:153 [Timer-0] Connect to http://drazzy.com/package_drazzy.com_index.json, method=HEAD, request id=86FDEF1223A141E4
2021-01-12T12:25:03.569Z INFO c.a.u.n.HttpConnectionManager:157 [Timer-0] Request complete URL="http://drazzy.com/package_drazzy.com_index.json", method=HEAD, response code=206, request id=86FDEF1223A141E4, headers={Accept-Ranges=[bytes], null=[HTTP/1.1 206 Partial Content], Server=[Apache/2.2.34 (Amazon)], Content-Range=[bytes 0-102310/102311], ETag=["1ea-18fa7-5b8a8b457c91b"], Connection=[close], Last-Modified=[Tue, 12 Jan 2021 00:00:03 GMT], Content-Length=[102311], Date=[Tue, 12 Jan 2021 12:25:04 GMT], Content-Type=[application/json]}
..............................
`

I will mention that prior to opening this issue, I had installed ttf-ms-fonts from the AUR, and also installed OpenJDK Java 7, 8 and 11, setting them to default in turn - problem remained.

igivis7, 2021-01-22:
I have got the same issue (Majaro KDE). This also appeared after the latest global update of the system. But in my case, it only appears when I run downloaded Arduino IDE via 'sudo'. If I run it from my normal user or from root all is ok. Also no problem with Arduino IDE installed via pacman/pamac.

ajmbelford, 2021-01-31:
Had the same problem. after trying mods to the java file, severnaside's workaround: "In the IDE download folder there is a folder named 'java'. I renamed it to something different and launched the IDE - everything working ok. ..." was sufficient to make it usable for me. 
Arduino IDE 1.8.13
5.10.10-200.fc33.x86_64 (inside KVM/QEMU on 5.10.10-200.fc33.x86_64)

mr-vb, 2021-02-01:
I am running Fedora Core 33 (FC33) with Arduino 1.8.13.  I experienced the same issue with the Menu Bar missing as well as the drop down only showing the black arrows when i started Arduino using the Activities screen pinned apps or via the search option. 
I resolved the issue by going to the terminal window and running   SUDO ARDUINO.  Some messages went by and Arduino came up with the Menu Bar showing as it should.  I then shut down the app and restarted it again at the command line using just ARDUINO and all worked as well.  Then I started it again as I had been and the issue was resolved.  So, it appears that the SUDO run of Arduino may have fixed what was causing the issue.   HTH.

jstoja, 2021-02-01:
@facchinm When running with the JRE shipped with version 1.8.3, I get the following error message:

```
Gtk-Message: 22:40:58.924: Failed to load module "pk-gtk-module"
```

Although, when upgrading Java as suggested by @lefeverd , I could see that `xorg-x11-fonts-Type1` was installed at the same time. Maybe that's related?

For the next readers:

```
# cd in your arduino IDE extracted folder
sudo dnf install java-latest-openjdk
sudo alternatives --config java # select the number with java-latest-openjdk.x86_64
mv java{,_old}
```

kerauzen, 2021-02-07:
Same issue on FC33.x86_64
with arduino 1.8.13 and  1.8.14
When sudoing arduino , the menu font is ok 
But it do not solve the issue with current user

and changing java with @lefeverd  proposal  does not solve the problem too

i ve got the message 
Failed to load module "pk-gtk-module"  ...

kerauzen, 2021-02-07:
By mv arduino/java to arduino/old-java, the menu are OK

CreaValix, 2021-02-25:
Same on Tumbleweed. @kerauzen , thanks, moving away the internal Java solves it.

Eontech, 2021-03-28:
On Fedora 33: with a local install (ie. No root, No sudo)

Downloaded the latest openjdk from: https://github.com/AdoptOpenJDK/openjdk11-binaries/releases
extract to a local folder (eg. ~/dev/openjdk/latest)
cd  ~/.local/.../arduino-1.8.13
mv java{,_old}
ln -s (extracted openjdk-folder ^^) java

It works fine after that.

echd, 2021-06-18:
Same issue on ArchLinux with both gtk2 and gtk3 packages installed, Arduino IDE 1.8.13 (64-bit). Uninstalled gtk2 package and problem was resolved.

It might be that the bundled Java gets confused about which L&F to use. Full support for mixed GTK2, GTK3 environments wasn't ironed out until JDK 9 with [JEP 283](https://openjdk.java.net/jeps/283) which set the default to GTK2 for all the various components (AWT/Swing, JavaFX, SWT). Oracle backported a pair of patches to use GTK3 as default L&F for [client-libs](https://bugs.java.com/bugdatabase/view_bug.do?bug_id=JDK-8222496) and [javafx](https://bugs.java.com/bugdatabase/view_bug.do?bug_id=JDK-8222497) in its 8u implementation but maybe the bundled implementation doesn't have similar patches and so experiences an "application failure" as mentioned in JEP 283.

belimawr, 2021-06-21:
Same issue here, ArchLinux (up to date as of today), I3 with a basic custom installation, renaming the `java` folder works.

Arduino 1.8.15 (downloaded today from arduino.cc)
Logs from running it from the terminal with arudino's java (menus do not work), I added a `set -ex` at the top of the `arduino` script to have more details:
```
+++ readlink -f -- ./arduino
++ dirname -- /home/belimawr/Software/arduino-1.8.15/arduino
+ APPDIR=/home/belimawr/Software/arduino-1.8.15
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-awt-util-1.8.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-awt-util-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-bridge-1.8.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-awt-util-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-bridge-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-codec-1.8.jar
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-awt-util-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-bridge-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-codec-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-css-1.8.jar                                                                                             
+ for LIB in "$APPDIR"/java/lib/rt.jar "$APPDIR"/java/lib/tools.jar "$APPDIR"/lib/*.jar
+ CLASSPATH=:/home/belimawr/Software/arduino-1.8.15/java/lib/rt.jar:/home/belimawr/Software/arduino-1.8.15/java/lib/tools.jar:/home/belimawr/Software/arduino-1.8.15/lib/apple.jar:/home/belimawr/Software/arduino-1.8.15/lib/arduino-core.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-anim-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-awt-util-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-bridge-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-codec-1.8.jar:/home/belimawr/Software/arduino-1.8.15/lib/batik-css-1.8.jar:/home/belima [... cut ...]
extra/jdk-openjdk 15.0.2.u7-1 [installed]
extra/jre-openjdk 15.0.2.u7-1 [installed]
extra/jre-openjdk-headless 15.0.2.u7-1 [installed]
extra/js68 68.12.0-1 [installed]
extra/js78 78.10.1-1 [installed]
extra/qt5-declarative 5.15.2+kde+r24-1 (qt qt5) [installed]
extra/qt5-webchannel 5.15.2-1 (qt qt5) [installed]
extra/webkit2gtk 2.32.0-2 [installed]
community/geos 3.9.1-1 [installed]
community/mujs 1.1.2-1 [installed]
community/nodejs-lts-dubnium 10.24.0-2 [installed]
```

I can provide more details if needed.

bobatlcpmbdx, 2021-07-30:
Having the same issues withe fedora 33 and arduino.
I've tried renaming the java directory but that resulted in no IDE launch at all from the command line both as user and root AND returns the following

bob@fred11 arduino-1.8.15]$ arduino
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/bob/.arduino15
java.lang.UnsatisfiedLinkError: no splashscreen in java.library.path: [/home/bob/arduino-1.8.15-linux64/arduino-1.8.15/lib, /usr/java/packages/lib, /usr/lib64, /lib64, /lib, /usr/lib]
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2670)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:830)
	at java.base/java.lang.System.loadLibrary(System.java:1873)
	at java.desktop/java.awt.SplashScreen$1.run(SplashScreen.java:134)
	at java.desktop/java.awt.SplashScreen$1.run(SplashScreen.java:132)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.desktop/java.awt.SplashScreen.getSplashScreen(SplashScreen.java:131)
	at processing.app.Base.<init>(Base.java:246)
	at processing.app.Base.main(Base.java:150)
bob@fred11 arduino-1.8.15]$

bobatlcpmbdx, 2021-07-30:
OK. so I have SOLVED MY version of the problem as above.
It appeared to me that the ardino launcher was not finding the correct version of java.
I installed Arduino using dnf in Fedora 33
This produced the same error as the version downloaded from Github I was using as above.
I had earlier installed java-latest-openjdk.x86_64 (/usr/lib/jvm/java-16-openjdk-16.0.1.0.9-4.rolling.fc33.x86_64/bin/java), renamed java directory in arduino, and had run 'sudo alternatives --config java-latest-openjdk-1:16.0.1.0.9-4.rolling.fc33.x86_64'
This had not worked probably because I have little understanding of where to correctly direct the command so that arduino could find java to launch the IDE
So started to look at re-installing java - not a good idea! However I did find a way to see which versions of java were on my machine and which was being used and it was not the latest version I had installed earlier.
I had seached in google for "installing java on fedora 33" and was directed to the Fedora Docs page for installing java (https://docs.fedoraproject.org/en-US/quick-docs/installing-java/)
page down to "Switching between Java Versions" and there is a clear explanation how to use the alternatives command. So this is what I did.

[bob@fred11 ~]$ sudo alternatives --config java
[sudo] password for bob: 

There is 4 program that provides 'java'.

  Selection    Command
-----------------------------------------------
*+ 1           java-11-openjdk.x86_64 (/usr/lib/jvm/java-11-openjdk-11.0.11.0.9-4.fc33.x86_64/bin/java)
   2           /usr/java/jre1.8.0_271-amd64/bin/java
   3           java-1.8.0-openjdk.x86_64 (/usr/lib/jvm/java-1.8.0-openjdk-1.8.0.292.b10-4.fc33.x86_64/jre/bin/java)
   4           java-latest-openjdk.x86_64 (/usr/lib/jvm/java-16-openjdk-16.0.1.0.9-4.rolling.fc33.x86_64/bin/java)

Enter to keep the current selection[+], or type selection number: 4
[bob@fred11 ~]$ arduino

BINGO it works now with menus

belimawr, 2021-11-01:
I've recently done a full OS reinstall on my laptop, download the latest version (1.8.16), but the bug and solution still the same. Renaming/deleting the `java` folder solves the problem.

The jdk I'm using:
```
$ java --version
openjdk 17 2021-09-14
OpenJDK Runtime Environment (build 17+35)
OpenJDK 64-Bit Server VM (build 17+35, mixed mode)
```

BRUXB32, 2021-11-05:
Renamed that folder and everything is fixed, i love linux!

dizzwold, 2021-11-06:
Hi Guy's,

 Are you installing and moving the ide before changing the Java file name?

belimawr, 2021-11-06:
> Hi Guy's,
> 
> Are you installing and moving the ide before changing the Java file name?

I download it directly from the website and extract the files somewhere on my home directory.

BRUXB32, 2021-11-08:
> Hi Guy's,
> 
> Are you installing and moving the ide before changing the Java file name?

dizzwold, I downloaded it from the page, and although not having moved it I think It won't really affect as far as you rename the "java" folder with another name, good luck!

djhunter67, 2023-02-13:
> sudo alternatives --config java

> Renamed that folder and everything is fixed, i love linux!

I rename the folder from `~/Downloads/arduino-1.8.19/java` to `~/Downloads/arduino-1.8.19/not_java`

Everything worked after that!

sptpower, 2024-11-13:
help pls. [image]

sptpower, 2024-11-13:
Solution to my problem -- install  need (openjdk-17-jdk)  sudo apt-get install openjdk-17-jdk.. [image]