arduino/Arduino #11153: java.lang.NullPointerException

bailez, 2021-01-11:
Installed Arduino directly from pacman:
arduino 1:1.8.13-2
arduino-builder 1.5.4-1
arduino-ctags 5.8_arduino11-4

My system:
LSB Version:	1.4
Operating System: Arch Linux
            Kernel: Linux 5.9.14-arch1-1
      Architecture: x86-64

Gnome:
<gnome-version>
 <platform>3</platform>
 <minor>38</minor>
 <micro>3</micro>
 <distributor>Arch Linux</distributor>
 

Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/bailez/.arduino15
java.lang.NullPointerException
	at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
	at processing.app.Base.<init>(Base.java:286)
	at processing.app.Base.main(Base.java:150)

facchinm, 2021-01-12:
Hi @bailez ,
Arch IDE packaging is community based (maintained by @NicoHood ). You probably miss a dependency (like `avr-core`, even if that would be a regression).
In any case, try downloading and running the ide from here https://www.arduino.cc/en/software . If it works, the bug should be moved to https://bugs.archlinux.org/?project=5&string=arduino

NicoHood, 2021-01-12:
That issue is tracked here, and still not fixed by arduino:
https://github.com/arduino/Arduino/issues/10580

facchinm, 2021-01-12:
Thanks Nico, closing this one as duplicate and keeping track in the other.
It would be cool if someone could bisect the IDE to understand when this regressed (my gut feeling is that the menu "compression" patch is missing a nullpointer check)

closed 2021-01-12 by facchinm

bailez, 2021-01-12:
Thanks guys, installing the AVR core fixed it!