arduino/Arduino #11161: Please add AskSin++ Library

pa-pa, 2021-01-13:
Please add the library to the library manager

https://github.com/pa-pa/AskSinPP

per1234, 2021-01-13:
Hi @pa-pa. Thanks for your interest in contributing to the Arduino Library Manager index.

In order to be accepted into the index, libraries must comply with some additional requirements. One of these is a prohibition against containing .exe files. the AskSinPP contains exe files here:
https://github.com/pa-pa/AskSinPP/tree/V5.0.0/bootloader/avr/srecord
These files would need to be removed from the repo or branch before the library can be added. You might consider only storing the source code in the repository and using GitHub release assets to provide compiled binaries associated with each release of the project. Other options would be to store them in a separate repository or branch.

Once that's done, either create a new tag/release or else redo the existing tag and then comment here to let us know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

pa-pa, 2021-01-14:
Sorry, that I missed this. The executable files are removed from master branch and from the tag.