arduino/Arduino #11162: Library folder naming in library manager vs github download

lemmingDev, 2021-01-14:
When downloading my library from the Github release page, eg: https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.8

The library inside the zip file is in the folder "ESP32-BLE-Gamepad-0.1.8", and people ordinarily rename it to "ESP32-BLE-Gamepad"

If however they download the library from the library manager, it installs into the folder "ESP32_BLE_Gamepad"

So, if the have been downloading the library from github, and then start using the library manager, they'll have 2 version of the library installed, and it seems the older github version will be used by Arduino.

This is confusing for users as the assume the latest version the installed from the library manager will be the one used.

I think this inconsistent folder naming is because the library.properties file lists the name of the library with spaces (name=ESP32 BLE Gamepad)

So, as a test, what I've done is changed the name property in the library.properties file to include dashes (name=ESP32-BLE-Gamepad), which will hopefully force the library to be extracted into a folder called ESP32-BLE-Gamepad which is what I hope for. https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.9 --> this version is not showing in the library manager. Does it take some some, or is it not showing because of the name change?

QUESTIONS: 
1. Will this change do what I want it to do?
2. Will the library need to be removed form the library manager under the old name and then re-added the the library manager under this new name, or will it be handled without doing that?

**How to reproduce:**
- Download an older release of the library and install it to the libraries folder, removing the version number from the folder name https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/0.1.4
- Install the latest version from the library manager
- Then there will be 2 versions of the library installed, but the older one will be the one picked up by Arduino (4 examples from the examples menu in 0.1.4 instead of 6 for the latest version)

per1234, 2021-01-14:
> this version is not showing in the library manager though. Does it take some some, or is it not showing because of the name change?

The latter. Because it is used as the key, libraries are locked to the library.properties `name` value they had when they were submitted to Library Manager. Any release with a different name is rejected by the indexer. But library maintainers are welcome to submit issues here requesting a name change (which is a manual process). We can consider this issue to be the request for the library name change if you like .

>Will this change do what I want it to do?

This change will result in the library being installed to a folder named `ESP32-BLE-Gamepad` . Since library folder names are not allowed to contain spaces, Library Manager replaces any spaces with `_` when creating the library installation folder. This is the only character substitution that is done, so if you don't like the `_` you can replace the spaces  in the `name` value yourself with the supported character (or lack thereof) of your preference, as you have done.

lemmingDev, 2021-01-14:
Thanks for that info.

So, can I please request my my original library be taken off the manager, and the new one be put on from this thread, or should I create another one?

per1234, 2021-01-14:
>Thanks for that info.

You're welcome.

>should I create another one?

This issue will be sufficient. It will be taken care of while processing the next batch of Library Manager index submissions.

lemmingDev, 2021-01-14:
Much appreciated

Might confuse a few early adopters of the library, but will be best in the long run...

lemmingDev, 2021-01-15:
Oh, and while I've got you, how long does it generally take for a library update on github to be reflected in the arduino library manager?

lemmingDev, 2021-01-28:
@per1234 

Sorry to bother you, but it has been 14 days and doesn't look like the issue still hasn't been resolved.

Happy to create a new issue if needed

Thanks heaps

per1234, 2021-01-28:
Hi @lemmingDev. Sorry for the delay. I have edited the issue title to make the request more clear.

Unfortunately, nobody ever bothered to create a dedicated issue label to use for the thousands of Library Manager submissions, so it's difficult to distinguish feature requests or support requests for Library/Boards Manager itself from the requests relating to the index submissions or updates when doing a quick skim unless it's clear from the issue title.

cmaglie, 2021-01-29:
> but it has been 14 days and it looks like the issue still hasn't been resolved.

@lemmingDev for some reason this issue went out of my radar.
Now it's done, sorry for the long wait, and thanks for your patience.

closed 2021-01-29 by cmaglie

lemmingDev, 2021-01-29:
@cmaglie  @per1234 
No apologies needed. I really appreciate the support you provide to the community.
Thanks so much for sorting the issue and keep up the great work