arduino/Arduino #11172: Moving window to a different display crashes IDE (Mac)

OliB150, 2021-01-17:
On MacOS Catalina 10.15.7 when moving any Arduino IDE window to a different display, the IDE instantly crashes.
Running Arduino IDE 1.8.12.

Not really sure what else I can add, that seems to be it, but its pretty repeatable so if more info is required, I can do it again (I'll make sure I have saved my files this time though, that was a bad mistake...)

per1234, 2021-01-17:
Closing as duplicate of https://github.com/arduino/Arduino/issues/9764

This bug has already been fixed in https://github.com/arduino/Arduino/pull/10234 but you're using an outdated version of the Arduino IDE from before the fix. Please update to Arduino IDE 1.8.13.

closed 2021-01-17 by per1234

OliB150, 2021-01-17:
Ah, apologies, I did search but only for open issues so I didn't spot that. I did check that I have "Check for updates on startup" enabled (which I do) and it hasn't prompted me to update at all, so I assumed I was already running the latest version. 

Thanks for the quick turnaround though!