arduino/Arduino #11177: Error loading WakaTime tool

gandarez, 2021-01-18:
I've deloped a tool for Processing IDE and as I read it supports tools from Processing IDE. I also noticed it inherited the tools system from Processing [here](https://forum.arduino.cc/index.php?topic=578486.0). No logs are generated at `~/Library/Arduino15/logs` nor any detailed information is thrown. The IDE shows the splash screen and after few seconds simply shuts downs.

Below is the error I could see when running via command line:
```
java.lang.AbstractMethodError
	at processing.app.Editor.addTools(Editor.java:885)
	at processing.app.Editor.buildToolsMenu(Editor.java:732)
	at processing.app.Editor.buildMenuBar(Editor.java:544)
	at processing.app.Editor.<init>(Editor.java:291)
	at processing.app.Base.handleOpen(Base.java:889)
	at processing.app.Base.handleOpen(Base.java:875)
	at processing.app.Base.handleOpen(Base.java:871)
	at processing.app.Base.handleNew(Base.java:810)
	at processing.app.Base.<init>(Base.java:498)
	at processing.app.Base.main(Base.java:150)
```

[Here](https://github.com/devgianlu/processing-wakatime) is the tool I used.

gandarez, 2021-09-05:
Anyone can help me with this?

per1234, 2021-09-05:
Hi @gandarez. Thanks for your interest in this open source project.
The Arduino IDE was forked from the Processing IDE many years ago and the two code bases have evolved significantly since that time, with no effort made to keep them in sync. So it's very unlikely that a Processing IDE tool will work out of the box with Arduino IDE.

There is a minimal demonstration tool that is intended to help people learn to write tools here:
https://github.com/arduino/Arduino/blob/master/build/shared/tools/Mangler/src/Mangler.java
You can find some other tools to study at the forum post you linked.

Feel free to post on the Arduino Forum if you have any questions:
https://forum.arduino.cc/

closed 2021-09-05 by per1234