arduino/Arduino #11178: New Library: BigFont02 + BigFont02_I2C

telleropnul, 2021-01-19:
[https://github.com/telleropnul/BigFont02](https://github.com/telleropnul/BigFont02)
[https://github.com/telleropnul/BigFont02_I2C](https://github.com/telleropnul/BigFont02_I2C)

telleropnul, 2021-01-20:
Arduino IDE > Tools > Manage Libraries... > search for "BigFont02" > 0 hits after 24 hours.

Action may be required.

per1234, 2021-01-20:
Hi @telleropnul 
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

cmaglie, 2021-01-21:
Done.

closed 2021-01-21 by cmaglie