arduino/Arduino #11182: [Library Manager] Please add MeteoLabBeacon.

vladimir-chukin, 2021-01-19:
Please add the MeteoLabBeacon library to the library manager:

https://github.com/meteolab/MeteoLabBeacon

Arduino library for Bluetooth Low Energy (BLE) environmental sensor communication using the nRF24L01+

per1234, 2021-01-19:
Hi @vladimir-chukin. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

vladimir-chukin, 2021-01-19:
Thank you for the detailed information. Release is created.

cmaglie, 2021-01-21:
Done.

closed 2021-01-21 by cmaglie