arduino/Arduino #11186: [Library Manager] Jeager-One Library

mchwalfajar, 2021-01-20:
**Please add**: https://github.com/mchwalfajar/jeager-one

About this library:
This library is used to transmit sensor data with LoRa, WiFi, and GSM connections

Thanks,
mchwalfajar

per1234, 2021-01-20:
Hi @mchwalfajar. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

mchwalfajar, 2021-01-20:
hi @per1234 done, thanks

cmaglie, 2021-01-21:
Done.

closed 2021-01-21 by cmaglie