arduino/Arduino #11190: [Library Manager] ESP8266RTTTLPlus Library

StarsoftAnalysis, 2021-01-21:
Hello

I've created a new and improved library for parsing and playing RTTTL melodies at:

https://github.com/StarsoftAnalysis/ESP8266RTTTLPlus

Please can you add it to the Arduino library?

per1234, 2021-01-21:
Hi @StarsoftAnalysis. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

StarsoftAnalysis, 2021-01-21:
Updated to version 0.1.2.

per1234, 2021-01-21:
@StarsoftAnalysis the repository still doesn't have any releases or tags:
https://github.com/StarsoftAnalysis/ESP8266RTTTLPlus/tags
Did you maybe make the tag in your local clone and then forget to push it to GitHub?

StarsoftAnalysis, 2021-01-21:
Thank you for your patience and quick replies.  I hadn't understood the need to use take extra steps to push tags.  I hope it's right now.

Cheers,

Chris

cmaglie, 2021-01-29:
Done.

closed 2021-01-29 by cmaglie