arduino/Arduino #11191: Update to Webbino library do not get picked up by Library Manager

SukkoPera, 2021-01-21:
I have recently updated my [Webbino](https://github.com/SukkoPera/Webbino) library and released a few new versions but they weren't picked up by the [LIbrary Manager](https://www.arduino.cc/reference/en/libraries/webbino/), which is still stuck at an old version.

This is probably due to some reorganization I've done to the directory structure of the repository.

Could this please be fixed? I am going to release 1.0.0 soon and it would be a pity to miss it :).

per1234, 2021-01-21:
Ah, I just saw the linkified text in your report points to the generated library reference page on arduino.cc:
https://www.arduino.cc/reference/en/libraries/webbino/
Which does indeed miss these versions:
- `0.9.95`
- `0.9.96`
- `0.9.97`

So it seems to me that Library Manager is working fine and this is really a report of a bug in the library reference page generation system. We had a previous report of the same thing here: https://github.com/arduino/Arduino/issues/10976

@alranel I see you commented on https://github.com/arduino/Arduino/issues/10976 that that issue would be fixed, but there hasn't been an update since then. What is the status of this issue?

SukkoPera, 2021-01-21:
Oh, sorry then. some friends reported some oddities to me and I thought it wouldn't show up in the LM. Checking that page convinced me of that. Actually I could see it, but I thought it was only because of course I have local development versions.

It would be great if you could fix the page issue too :)