arduino/Arduino #11192: Created a new Arduino Library for Displaying Numbers and Letters on 7-segment display

alikabeel, 2021-01-22:
Hi,
I have created a new library that allows displaying all alphanumeric characters from (A-Z & 0-9). Can you please add it to the Arduino library manager?
Github Repo of Library: https://github.com/alikabeel/Letters-and-Numbers-Seven-Segment-Display-Library
Best Regards,
Ali Kabeel

alikabeel, 2021-01-28:
Hello any updates on this issue?

per1234, 2021-01-28:
Hi @alikabeel 
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

cmaglie, 2021-01-29:
Done.

closed 2021-01-29 by cmaglie