arduino/Arduino #11195: Error code in arduino ide

Demultiplexer1, 2021-01-23:
If someone know how to fix this please help me!

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

atharwa-24, 2021-04-09:
@Demultiplexer1 try extracting the file where its says missing Ref. this https://www.esp8266.com/viewtopic.php?p=14447#p14447

atharwa-24, 2021-04-09:
Its Included here : https://github.com/rogerclarkmelbourne/arduino-esp8266/archive/master.zip