arduino/Arduino #11196: Auto-select correct COM port

MMartin09, 2021-01-23:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
In most cases, a "default" user will only use a single Arduino at the time. In Windows, there are usually multiple open COM ports. 
My machine, for example, always has at least two ports. But I think it should be no problem to auto-select the port where the Arduino is connected. In fact, I already implemented this in some of my applications. An auto-selected COM port is a good thing in the IDE i think.

per1234, 2021-01-23:
There's a proposal to go at in in the opposite direction (select board based on port selection) here: https://github.com/arduino/Arduino/pull/7120.

Arduino Web Editor and Arduino Pro IDE do have this functionality already.

MMartin09, 2021-02-08:
Thank you. I think this could close the issue.

closed 2021-02-08 by MMartin09