arduino/Arduino #11197: Please add  this New library for Bot Control

paulpavish, 2021-01-23:
Opening this issue As mentioned in https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ on how to add a library to Arduino Library Manager.

The library to be included, link : https://github.com/paulpavish/MotorDriver_GuiNE_Bot/releases/tag/v1.0.0

Thankyou
Paul Pavish

paulpavish, 2021-01-24:
Hi, Thank you for responding. 
Just out of curiosity to begin my sessions, How long will it take for this library to get listed in the Arduino LIbrary Manager !?

per1234, 2021-01-24:
Hello @paulpavish. Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache.

paulpavish, 2021-01-24:
I understand @per1234 .
Thank you

cmaglie, 2021-01-29:
Done.

closed 2021-01-29 by cmaglie