arduino/Arduino #11206: [library manager] CAN (Adafruit Fork)

ladyada, 2021-01-25:
https://github.com/adafruit/arduino-CAN

this is a fork to add support for the SAME51, as https://github.com/sandeepmistry/arduino-CAN/pull/57 has not been merged and folks want to use the library manager to install

per1234, 2021-01-26:
Hi @ladyada. I spotted a couple issues:

- The [library.properties `name` value](https://github.com/adafruit/arduino-CAN/blob/master/library.properties#L1) contains the [prohibited characters](https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format) `(` and `)`.
- The library.properties `name` value at the time of the last tag was "CAN", which has already been taken by Sandeep's library. The indexer works from tags, so it won't recognize an updated `name` value if it isn't in the latest tag.

I think it will all be good if you update the `name` value and then create a tag. After doing that, please comment here so we'll know we can proceed.

ladyada, 2021-01-26:
ok updated name to remove parens and bumped to 1.0.1

cmaglie, 2021-01-29:
Done.

closed 2021-01-29 by cmaglie