arduino/Arduino #11212: [Library Manager] Please add the ESPCanary Library to the Library Manager

mudmin, 2021-01-26:
Please add the ESPCanary library. It turns an ESP8266 or ESP32 into a honeypot FTP server that fires off an opencanary.org security token or hits the webhook of your choosing.  You can also store legitimate files and other tokens on the SPIFFS file system as well as change the MAC address of the ESP to make it look like specific hardware.  It looks like a Synology box by default.

per1234, 2021-01-27:
Hi @mudmin. Thanks for your interest in contributing to the Arduino Library Manager index.

I spotted a problem with your library. The library.properties `url` field is empty:
https://github.com/mudmin/ESPCanary/blob/1.0.2/library.properties#L8
```
url=
```
Please add a URL to this field and then either create a new release/tag or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

More information:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

mudmin, 2021-01-27:
Thank you for catching that.  I fixed it and updated the repo.   I appreciate your time.

cmaglie, 2021-01-29:
I've added the library to the indexer but there is a problem, you must define the `author` field in the library.properties.:

```
JOB 3657 - 2021/01/29 15:08:08 Checking out tag: v1.0.3
JOB 3657 - 2021/01/29 15:08:08 Error syncing library: Error generating library from repo: 'author' field must be defined
```

To fix this:
1. edit library.properties and define the `author` field
2. commit & push
3. delete and redo the tag v1.0.3 on the new master

once done, our indexer will automatically pick up and publish the library within an hour.
I'm going ahead and close this issue, feel free to comment here if you need further assistance.

closed 2021-01-29 by cmaglie

mudmin, 2021-01-30:
Thank you for your help @cmaglie 
I just have one last question.  If I update the library do I need to come back here and do anything or do I just do a new library properties, tag and release?

per1234, 2021-01-30:
The Library Manager indexer automatically picks up any new tag you make as long as it meets all the requirements listed here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

It does take some time for the indexer job to run and the index update to propagate through the CDN cache, I'd guess it will always be less than a day before the new release becomes available for installation via Library Manager though.

mudmin, 2021-01-30:
Fantastic. I can't tell you how much I appreciate your help on this and all you're doing for the maker community.