arduino/Arduino #11213: Please add Vcc library to Library Manager

felias-fogg, 2021-01-26:
Hi,

could ypu please add my Vcc library to the Library Manager: https://github.com/felias-fogg/Vcc

Cheers,
Bernhard

per1234, 2021-01-27:
Hi @felias-fogg. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

felias-fogg, 2021-01-27:
Sorry. Now I've created a release.

cmaglie, 2021-01-29:
Done.

closed 2021-01-29 by cmaglie