arduino/Arduino #11216: Mac OSX Catalina not show lib examples

RodrigoDornelles, 2021-01-27:
i'm using a macbook with the latest version of mac os, and i can't list the examples created by third party libraries.

[image]

per1234, 2021-03-29:
Hi @RodrigoDornelles. Thanks so much for reporting this.

In order to assist us with the investigation, please try this quick experiment of installing an arbitrary 3rd party library via Library Manager:

1. From the Arduino IDE's menus, select **Sketch > Include Library > Manage Libraries...**
2. Wait for the update to finish.
3. In the "Filter your search..." field, type "fastled"
4. Press the "Enter" key.
5. From the list of search results, click on "FastLED by Daniel Garcia".
6. Click the "Install" button.
7. Wait for the installation to finish.
8. Click the "Close" button.

Now open the **File > Examples** menu. Do you see "FastLED" in the menu?

per1234, 2021-05-15:
Closing due to lack of response.

@RodrigoDornelles  if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-05-15 by per1234

RodrigoDornelles, 2021-05-15:
Sorry for the delay, I was without a computer, but finally when I fixed my macbook, this problem no longer occurs.