arduino/Arduino #11217: IDE 1.8.13 for ArduBlock

caudld, 2021-01-27:
If I click "upload" button in Ardublock will error. The error messages:

Arduino Version: unknown
Exception in thread "AWT-EventQueue-0" java.lang.NoSuchMethodError: processing.app.Editor.setText(Ljava/lang/String;)V
	at com.ardublock.ArduBlockTool.didGenerate(ArduBlockTool.java:62)
	at com.ardublock.core.Context.didGenerate(Context.java:253)
	at com.ardublock.ui.listener.GenerateCodeButtonListener.actionPerformed(GenerateCodeButtonListener.java:174)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

I saw that this issue was reported in 2016 on this thread: https://github.com/arduino/Arduino/issues/5454 and I wondered if the same thing was going on now.

per1234, 2021-01-28:
Hi @caudld. Which version of ArduBlock are you using? Where did you get it from?

caudld, 2021-01-28:
Hi @per1234, I'm not sure which version of ArduBlock I downloaded, but I downloaded it from the ArduBlock website: http://blog.ardublock.com/engetting-started-ardublockzhardublock/
The title of the file it led me to download was ardublock-all-20130712.jar .

per1234, 2021-01-28:
OK, that's the problem. It seems that the project is not well maintained any more. You are getting a version from 2013, long before the fix was made for compatibility with the modern Arduino IDE. Since there haven't been any production releases of Ardublock with the fix, you'll need to use the beta version downloaded from here:
https://github.com/taweili/ardublock/releases/tag/beta-20171028
There are some installation instructions in the readme:
https://github.com/taweili/ardublock/tree/beta-20171028#installation

closed 2021-01-28 by per1234

caudld, 2021-01-28:
Thank you so much!