arduino/Arduino #11233: ___REMOVE___/esptool.py error uploading to ESP8266/32

RaSTuS26, 2021-02-01:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
IDE 1.8.13

Whenever I try and upload a sketch to ESP8266 or ESP32 I get the following error:
```
Sketch uses 396341 bytes (6%) of program storage space. Maximum is 6553600 bytes.
Global variables use 17652 bytes (0%) of dynamic memory, leaving 4504332 bytes for local variables. Maximum is 4521984 bytes.
python ___REMOVE___/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/rob/.arduino15/packages/m5stack/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /home/rob/.arduino15/packages/m5stack/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /tmp/arduino_build_801573/ServoKit180_360.ino.bin 0x8000 /tmp/arduino_build_801573/ServoKit180_360.ino.partitions.bin 
python: can't open file '___REMOVE___/esptool.py': [Errno 2] No such file or directory
An error occurred while uploading the sketch
```

I've no idea what the problem is, I have the latest esptool.py installed and flashing .bin files manually or through vscode works fine.  It's only in Arduino IDE I have the problem.

RaSTuS26, 2021-02-01:
Worked it out.  I was using board "M5Stack Arduino/M5Stack-FIRE", when I changed to "ESP32 Arduino/M5Stack-Fire" things started working again.  I tries to delete "M5Stack Arduino" from Boards Manager and kept getting "java.lang.NullPointerException" errors.  So I renamed "packages/M5stack" adding a .bak extension, then reinstalled it from Board Manager, now it works fine, so seems something was corrupt in the old tree.

RaSTuS26, 2021-02-01:
Seems fixed now

closed 2021-02-01 by RaSTuS26

FrightRisk, 2023-03-07:
Thanks for reporting back