arduino/Arduino #11237: Add library - uECG device support

the-3d6, 2021-02-03:
Please add a library that allows simple data retrieval from uECG device: https://github.com/ultimaterobotics/uECG_library

per1234, 2021-02-04:
Hi @the-3d6. Thanks for your interest in contributing to the Arduino Library Manager index.

I spotted an issue with your library that will block it from being added to the index. You need to provide a URL via the `url` field in the [library.properties file](https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format):
https://github.com/ultimaterobotics/uECG_library/blob/main/library.properties#L8
```
url=
```
You can always just put the repository URL in that field if there isn't any other URL you prefer to direct the users to for more information about the library.

After that, create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

Once those things are done, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

the-3d6, 2021-02-04:
Done (I hope in the right way, never used tags before)

per1234, 2021-02-04:
Hi @the-3d6. The `url` value looks good now, but the respository still doesn't have any tags:
https://github.com/ultimaterobotics/uECG_library/tags
The Library Manager indexer works exclusively from Git tags, so only the tagged versions of the library are available for installation via Library Manager.

There are a couple of options to create a tag:

You can use Git directly to make a tag. Information about that here:
https://git-scm.com/docs/git-tag
If doing this, be sure to use `git push` to push the tag from your local clone of the repository to the remote repository on GitHub.

A simpler option is to use the GitHub web interface to create a release. Every GitHub release also creates a Git tag. You can learn about that here:
https://docs.github.com/en/github/administering-a-repository/managing-releases-in-a-repository

the-3d6, 2021-02-06:
After struggling a bit with tags, finally made a release via github web interface, now it should be good :)

cmaglie, 2021-02-09:
Done.

closed 2021-02-09 by cmaglie