arduino/Arduino #11245: Incorrect ICSP header pin numbering on Nano pinout diagram

per1234, 2021-02-05:
The pinout diagram downloaded from the "Download the full pinout diagram as PDF [here](https://content.arduino.cc/assets/Pinout-NANO_latest.pdf)." on the "Documentation" tab of the Arduino Nano product page:
https://store.arduino.cc/arduino-nano
has incorrect pin numbering on the ICSP header:
[image]

Pin 1 is at the bottom left corner of the header, as indicated by the white dot on the PCB silkscreen:
[image]
So the header pin numbering is off by 180 degrees.

CC: @FunzioneFabio

FunzioneFabio, 2021-02-08:
Thank you @per1234 for finding the error, I will fix this.

FunzioneFabio, 2021-02-08:
Pinout uploaded and updated online, you can check it here:
https://content.arduino.cc/assets/Pinout-NANO_latest.pdf

@per1234 I think we can close the issue.

per1234, 2021-02-08:
Looks good. Thanks @FunzioneFabio !

closed 2021-02-08 by per1234