arduino/Arduino #11249: Can i change the name of my existing library ?

paulpavish, 2021-02-05:
My 'MotorDriver_GuinEBot' library was added to the library manager a week ago. Can i change the name of the library to just 'GuinEBot' but the header files will be the same. !?

per1234, 2021-02-05:
Yes, you can change the library name. You'll need to do that following:

1. Change the `name` value in library.properties to the new name: https://github.com/paulpavish/MotorDriver_GuiNE_Bot/blob/main/library.properties#L1
2. Update the `version` value in library.properties to the version you will be using for the release of the newly named library.
3. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.
4. Wait for the entry for your library in the index to be manually updated.

paulpavish, 2021-02-07:
Ok. what happens if i change the library repository name and then release a new version to the same library with different name and version 1.0.1 , will the library still be updated?

per1234, 2021-02-07:
Even though GitHub automatically sets up a redirect when you change the repository name, we should also change the URL in the Library Manager index just to be safe if you change the repository name. So please let us know if you do that.

paulpavish, 2021-02-08:
Ok. This is the already given repository link 'https://github.com/paulpavish/MotorDriver_GuiNE_Bot' . Renamed to 'https://github.com/paulpavish/GuiNE_Bot_Arduino_Library'

paulpavish, 2021-02-09:
Still pending update.....

per1234, 2021-02-09:
cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the request is processed is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

cmaglie, 2021-02-09:
Done.

closed 2021-02-09 by cmaglie