arduino/Arduino #11250: Please add library to Library Manager

EasyG0ing1, 2021-02-06:
Please add this library to the Library Manager:

https://github.com/EasyG0ing1/BlockNot

Thank you,

Mike Sims

per1234, 2021-02-06:
Hi @EasyG0ing1. Thanks for you interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

EasyG0ing1, 2021-02-06:
OK, the release has been created, sorry about that. :-)

EasyG0ing1, 2021-02-07:
Is there something else I need to do to help this process?

per1234, 2021-02-07:
@EasyG0ing1 no, everything should be fine now. You only need to wait until your library is added to the index. This issue will be closed at that time and you'll get a notification from GitHub. After that, there will still be some hours delay for the indexing process to finish before the library will be available for installation via Library Manager.

cmaglie, 2021-02-09:
Done.

closed 2021-02-09 by cmaglie