arduino/Arduino #11256: serial monitor doesn't show anything

1Misaka, 2021-02-08:
[image]

The arduino serial monitor doesn't show anything.Although i use Serial.begin(115200) and Serial.println(),the monitor even show nothing.

1Misaka, 2021-02-08:
In another word,it seems like a white page.

per1234, 2021-02-08:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2021-02-08 by per1234

matthijskooijman, 2021-02-08:
Do you mean the serial monitor shows *nothing* at all? Not even an input box, baud rate dropdown, etc.?

1Misaka, 2021-02-08:
> Do you mean the serial monitor shows *nothing* at all? Not even an input box, baud rate dropdown, etc.? If those are *not* shown, then this might be a bug, otherwise, it's likely some problem in your setup to be handled as @per1234 suggests.

   Input box and baud rate are all exist.
yes,the monitor like a white page.