arduino/Arduino #11261: missing german coment in desktop file

freddii, 2021-02-10:
at the moment the desktop file for linux (https://github.com/arduino/Arduino/blob/master/build/linux/dist/desktop.template)
only shows an english comment.

It would be nice to add an german comment to it.

Comment[de]=Quelloffene Plattform zum Elektronik-Prototyping

dev-monishsolanki, 2023-09-20:
Can I work on this issue  #11261

per1234, 2023-09-20:
Hi @dev-monishsolanki. Thanks for your interest in contributing to this project!

The Arduino IDE 1.x project is no longer under active development so it is unlikely that resources would be allocated to reviewing, accepting, and releasing minor enhancements such as the one proposed here.

If you are interested in successfully making a contribution to the Arduino IDE project, please contribute to the actively developed Arduino IDE 2.x project. You can learn about that here:

https://github.com/arduino/arduino-ide/blob/main/docs/CONTRIBUTING.md#contributor-guide

Since you expressed an interest in working in localization specifically, I will also point out that there are opportunities to efficiently make significant contributions to Arduino IDE 2.x in that area. Information on that subject is available here:

https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/translation.md#translator-guide

dev-monishsolanki, 2023-09-20:
ok thanks