arduino/Arduino #11267: Arduino IDE crashes immediately upon opening - Mac OS 11.2.1

dcoukos, 2021-02-12:
Hi, I just downloaded the Arduino IDE. 
Every time I try to open the application on my Mac it crashes. Would love to provide more info but it happens immediately and I'm not sure if a log is generated or where it would be. 

Arduino IDE v1.8.13
Mac OS 11.2.1
MacBook Pro 13inch, Retina, late 2013

danielktdoranie, 2021-02-12:
Same for me. Using macOS Big Sur 11.2.1 (20D74)

MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)
2.7 GHz Quad-Core Intel Core i7
16 GB 2133 MHz LPDDR3

After a couple of attempts I can get Arduino IDE to open but when compiling code for a DigiSpark I get this error:

Arduino: 1.8.13 (Mac OS X), Board: "Digispark (Default - 16.5mhz)"

fork/exec /Users/ddoran/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++: bad CPU type in executable
Error compiling for board Digispark (Default - 16.5mhz).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Don't have this issue on my Windows 10 machine.

per1234, 2021-02-13:
@danielktdoranie that is a completely different issue. The problem is that the Digispark board's package is unmaintained and since the time it was created macOS has dropped support for 32 bit executables. There's nothing Arduino can do about that. You can learn more about it here: https://github.com/digistump/DigistumpArduino/issues/105 and probably resolve your problem by using this nice community fork that has been modernized:
https://github.com/ArminJo/DigistumpArduino

dcoukos, 2021-02-15:
I closed this issue, because I think it's the same as [#10964](https://github.com/arduino/Arduino/issues/10964)

closed 2021-02-15 by dcoukos