arduino/Arduino #11271: [Library Manager] Becoming manager of Keypad Library (was #3507)

Nullkraft, 2021-02-16:
My name is Mark Stanley and I'm one of the authors of the _Keypad Library_. Can you kindly (re)point the Library Manager to https://github.com/Nullkraft/Keypad

**Justification below this point**
It's kind of a long story but Chris--A established the link (issue #3507) between his repo and the LM. I've tried contacting him several times but he never got back to me and there's been no activity on his github in several years. There's an issue that needs an update, https://github.com/Nullkraft/Keypad/issues/12, that I would like to make happen.

I pushed the latest changes to my repo and tagged it v3.2.0 while also updating library.properties

per1234, 2021-02-16:
Hi @Nullkraft. There is a problem with your library's metadata which would prevent it being added to the Arduino Library Manager index.

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries. The `name` value of your library is currently "Arduino Keypad":
https://github.com/Nullkraft/Keypad/blob/master/library.properties#L1
```
name=Arduino Keypad
```

The `name` value is also relevant to your request that https://github.com/Nullkraft/Keypad replace the @Chris--A version of the library (https://github.com/Chris--A/Keypad). The `name` value is the unique identifier of each library in the index, so your library would need to use the same name as the one @Chris--A chose: "Keypad".

Once the `name` value issue is resolved, it will also be necessary for you to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Nullkraft, 2021-02-16:
Thanks and I think I got everything cleaned up. The name in the properties file is now Keypad and the tag version is v3.2.1

  https://github.com/Nullkraft/Keypad

Rather boneheaded mistake on my part. I read the requirements in the double-negative and thought that I was required to include the Arduino name.

cmaglie, 2021-02-17:
@Nullkraft @per1234 
We cannot change the ownership of a library without the consent of the author.

We cannot know if @Chris--A cares about it so we simply cannot do it (unless @Chris--A shows up and explicitly allow it of course). It's particularly unfortunate in this case, I know, but that's how it works.

@Nullkraft I must ask you to choose a name different from `Keypad` (and that did not begin with `Arduino*` :-).
Sorry for the hassles and thanks for your patience!

Nullkraft, 2021-02-17:
Hello Christian,

I appreciate the time you've taken to work on this but I still hope to retain the Keypad name. There's not too much pressure to make the change right this moment so I would like to spend a little more time trying to reach Chris--A. I found another web page of his and sent an email to that domain. We'll see if he responds first.

If you would like to close the ticket I can just open another one later if/when I make an progress.

Best regards,
Mark Stanley