arduino/Arduino #11274: [Library Manager] Please add these libraries for our device "Agrumino Lemon"

Bighoneypot, 2021-02-16:
Hello everyone and thank you for your work.

I ask you to add this library on Arduino for our "Agrumino Lemon" sensor.

Thank you.
If there are any problems with the library write me.

https://github.com/lifely-cc/agruminoLemon-libraries

per1234, 2021-02-16:
Hi @Bighoneypot. Thanks for your interest in contributing to the Arduino Library Manager index. I get a "Page not found" error from the link you shared:
https://github.com/lifely-cc/agruminoLemon-libraries
Is the repository set to "private" visibility? I do see this similarly named repository:
https://github.com/lifely-cc/Agrumino

Bighoneypot, 2021-02-16:
@per1234  Sorry now is public

Bighoneypot, 2021-02-17:
@per1234 The correct repository is https://github.com/lifely-cc/agruminoLemon-libraries

cmaglie, 2021-02-17:
Done.

closed 2021-02-17 by cmaglie

Bighoneypot, 2021-05-19:
@maglie Gday , I have a question.
I have update my libraries with a new release.
To use the new release on the Arduino libraries, do I need to ask for some authorization or is it all automatic? Thank you.

per1234, 2021-05-19:
Hi @Bighoneypot every release which meets all the requirements listed in [the FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) will be automatically added to the Library Manager index.

The job runs hourly but there might be some additional delays before the release is available for installation via Library Manager due to caching. Worst case scenario, I would say it should always show up within a day.

Bighoneypot, 2021-05-20:
Thank you @per1234