arduino/Arduino #11278: [Library Manager] Add new library

djaumann, 2021-02-17:
Could you please add this library to the library-manager:
https://github.com/Infineon/arduino-rgb-led-lighting-shield

Thanks a lot already and have a nice week!

BR,
Dominik

per1234, 2021-02-17:
Hi @djaumann. Thanks for your interest in contributing to the Arduino Library Manager index.

I get a "Page not found" error from that link. Is the repository visibility set to "public"? I do see a similarly named repository:
https://github.com/Infineon/RGB-LED-Lighting-Shield-XMC1202

djaumann, 2021-02-17:
Hey @per1234,

I'm sorry I totally forgot to make the repository public ...
Now it should be available :)

The other one you mentioned is a old one and will be archived soon.

Thank you already and sorry for the late response!

BR,
Dominik

cmaglie, 2021-02-17:
Done.

closed 2021-02-17 by cmaglie

djaumann, 2021-02-18:
Thank you for doing it so quickly!