arduino/Arduino #11282: Arduino Idle is Fezzes

AniketFasate27, 2021-02-18:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

When I try to save my code in arduino idle it freezes sometimes. So I have to go in task manager and then i have to close it. So every time i am try to save new code freezes. So please give me some solution on it.

PaulStoffregen, 2021-02-18:
Maybe OneDrive or other network filesystem in play?

AniketFasate27, 2021-02-19:
No , Nothing working in background.

feikname, 2021-02-19:
How long does it freeze for?

AniketFasate27, 2021-02-19:
Infinite time. To close it i have to go task manager then it will close.

AniketFasate27, 2021-02-23:
please help its happening several times. Now i am fustrated

PaulStoffregen, 2021-02-24:
How do you imagine anyone could help from only the meager info you've given?

I don't even see which version of Arduino you're using, or whether your computer is Windows, Linux or Macintosh, and which version.  We are (maybe helpful) strangers on the internet.  We can't see your computer, we can't see your screen, we can't know what other software you've installed, or any other info to try guessing what might be different from your computer and ours and every other where this problem does not happen.

I mean, I'd like to help you.  But if you're feeling frustrated, just trying to imagine how anyone who wants to help must feel when reading this no-info github issue?

AniketFasate27, 2021-02-24:
Thank you for your response.

per1234, 2021-03-17:
Closing due to lack of enough information to investigate.

@AniketFasate27 if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-03-17 by per1234

Electro-boy, 2022-12-16:
Even My IDE freezes when Saving new sketch or while opening new sketch.
IDE freezes and if i click on the window or anywhere on the arduino ide , it plays error sound and no response. I need to manually open task manager and close the application and re-open again to work.

1.8.19 is the Arduino IDE version.
Windows 10

No application running in background including ONEDRIVE.