arduino/Arduino #11283: "update your libraries" pop-up is annoying and prevent from typing.

macusking, 2021-02-18:
Hello Devs, 

First, thanks for your hard-work. Arduino is a great plataform.

I have a request: please, could you remove (or reimplement) the "upload your libraries" pop-up. It prevents me from typing and it's really annoying.

[image]

Thanks so much for you attention!

matthijskooijman, 2021-02-18:
IIRC you can disable the check for updates in the preferences, would that solve your issue?

macusking, 2021-02-18:
> your

Thank you so much for your time and attention.

Yes, I'll disable it. Thanks.

In the future releases, if I'm to suggest, I'd love to see this pop-up not interrupting the keyboard typing. I bet some new corners face the same too.

Keep up your great work. Arduino IDE rocks!

macusking, 2021-02-18:
> IIRC you can disable the check for updates in the preferences, would that solve your issue?

Thank you so much for your time and attention.

Yes, I'll disable it. Thanks.

In the future releases, if I'm to suggest, I'd love to see this pop-up not interrupting the keyboard typing. I bet some new corners face the same too.

Keep up your great work. Arduino IDE rocks!

matthijskooijman, 2021-02-18:
I recall that creating such a popup was already quite a challenge within the java environment we use, I suspect it might be hard to put it on top, but not steal focus. Nevertheless, it sounds like a good suggestion, so I'll retitle this issue to track it.

Note that there was originally another issue where the popup would cause the IDE window to be focused even it was not before, see #3809 (that's now solved and different issue, though).