arduino/Arduino #11295: Builder drops a stitch

sterretjeToo, 2021-02-22:
Please see https://forum.arduino.cc/index.php?topic=728959.0 where the issue was raised

Source code:
`
const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE HTML><html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="https://use.fontawesome.com/releases/v5.7.2/css/all.css" integrity="sha384-fnmOCqbTlWIlj8LyTjo7mOUStjsKC4pOpQbqyi7RrhN7udi9RwhKkMHpvLbHG9Sr" crossorigin="anonymous">
  <style>
    html {
     font-family: Arial;
     display: inline-block;
     margin: 0px auto;
     text-align: center;
    }
    h2 { font-size: 3.0rem; }
    p { font-size: 3.0rem; }
    .units { font-size: 1.2rem; }
    .dht-labels{
      font-size: 1.5rem;
      vertical-align:middle;
      padding-bottom: 15px;
    }
  </style>
  <script>
  var count=10;

var counter=setInterval(timerjs, 1000); //1000 will  run it every 1 second
  function timerjs()
{
  count=count-1;
  if (count <= 0)
  {
     clearInterval(counter);
     return;
  }

 document.getElementById("timerjs").innerHTML=count + " secs"; // watch for spelling
}
</script>
</head>
<body>
  <h2>Remote 1 temperature & humidity</h2>
  <p>
    <i class="fas fa-thermometer-half" style="color:#059e8a;"></i>
    <span class="dht-labels">Temperature</span>
    <span id="temperature">%TEMPERATURE%</span>
    <sup class="units">&deg;C</sup>
  </p>
  <p>
    <i class="fas fa-tint" style="color:#00add6;"></i>
    <span class="dht-labels">Humidity</span>
    <span id="humidity">%HUMIDITY%</span>
    <sup class="units">%</sup>
  </p>
  <p>
  Time to next sample: <span id="timerjs"></span>
  </p>
</body>
<script>
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("temperature").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/temperature", true);
  xhttp.send();
}, 10000 ) ;

setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("humidity").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/humidity", true);
  xhttp.send();
}, 10000 ) ;
</script>
</html>)rawliteral";

void setup()
{

}

void loop()
{

}
`

Content of C:\Users\sterretje\AppData\Local\Temp\arduino_build_600637\sketch\sketch_feb22a.ino.cpp 
`#include <Arduino.h>
#line 1 "C:\\Users\\sterretje\\AppData\\Local\\Temp\\arduino_modified_sketch_915464\\sketch_feb22a.ino"
#line 1 "C:\\Users\\sterretje\\AppData\\Local\\Temp\\arduino_modified_sketch_915464\\sketch_feb22a.ino"

const char index_html[] PROGMEM = R"rawliteral(
<!DOCTYPE HTML><html>
<head>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="https://use.fontawesome.com/releases/v5.7.2/css/all.css" integrity="sha384-fnmOCqbTlWIlj8LyTjo7mOUStjsKC4pOpQbqyi7RrhN7udi9RwhKkMHpvLbHG9Sr" crossorigin="anonymous">
  <style>
    html {
     font-family: Arial;
     display: inline-block;
     margin: 0px auto;
     text-align: center;
    }
    h2 { font-size: 3.0rem; }
    p { font-size: 3.0rem; }
    .units { font-size: 1.2rem; }
    .dht-labels{
      font-size: 1.5rem;
      vertical-align:middle;
      padding-bottom: 15px;
    }
  </style>
  <script>
  var count=10;

var counter=setInterval(timerjs, 1000); //1000 will  run it every 1 second
#line 27 "C:\\Users\\sterretje\\AppData\\Local\\Temp\\arduino_modified_sketch_915464\\sketch_feb22a.ino"
function timerjs();
#line 27 "C:\\Users\\sterretje\\AppData\\Local\\Temp\\arduino_modified_sketch_915464\\sketch_feb22a.ino"
  function timerjs()
{
  count=count-1;
  if (count <= 0)
  {
     clearInterval(counter);
     return;
  }

 document.getElementById("timerjs").innerHTML=count + " secs"; // watch for spelling
}
</script>
</head>
<body>
  <h2>Remote 1 temperature & humidity</h2>
  <p>
    <i class="fas fa-thermometer-half" style="color:#059e8a;"></i>
    <span class="dht-labels">Temperature</span>
    <span id="temperature">%TEMPERATURE%</span>
    <sup class="units">&deg;C</sup>
  </p>
  <p>
    <i class="fas fa-tint" style="color:#00add6;"></i>
    <span class="dht-labels">Humidity</span>
    <span id="humidity">%HUMIDITY%</span>
    <sup class="units">%</sup>
  </p>
  <p>
  Time to next sample: <span id="timerjs"></span>
  </p>
</body>
<script>
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("temperature").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/temperature", true);
  xhttp.send();
}, 10000 ) ;

setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    if (this.readyState == 4 && this.status == 200) {
      document.getElementById("humidity").innerHTML = this.responseText;
    }
  };
  xhttp.open("GET", "/humidity", true);
  xhttp.send();
}, 10000 ) ;
</script>
</html>)rawliteral";

void setup()
{

}

void loop()
{

}
`
Please note the inclusing of #line in the output that does not exists in the ino file

sterretjeToo, 2021-02-22:
Additional question: How the hell do I get properly formatted code in github issues?

per1234, 2021-02-22:
@sterretjeToo use [Markdown code fencing](https://guides.github.com/features/mastering-markdown/#examples), three backticks before and after the code:
````
```
code goes here
```
````
You can even get fancy and specify the language to get syntax highlighting!
````
```cpp
C++/Arduino code goes here
```
````