arduino/Arduino #11296: Arduino driver library for LCLV like Adafruit 3627 and 3330

aggregate, 2021-02-22:
Hi!
Please consider adding my library for LCLV (liquid crystal light valves) like Adafruit 3627 and 3330

I have tagged the release v1.0.0

link to github:
https://github.com/aggregate/LCLV

This is my first Arduino library, so suggestions are welcome.  Thanks.

aggregate, 2021-02-22:
Sorry about the quick update, but I'm still new to this. There's now a v1.0.1 version simplifying to just one constructor.

cmaglie, 2021-03-03:
I added the library to the indexer but there is a problem, in your library.properties the `depends` field is not set correctly:

```
depends=analogWrite() is needed for proportional control of opacity
```

this field is not free text, it must be used to list other libraries that are required by this one to work properly.
If no other libraries are required (as I think is your case) you should just remove the field entirely.

To fix this:

1. edit your `library.properties` and remove the `depends` field
2. commit & push
3. delete and redo the v1.0.1 tag on the new `main`

once done the library will be automatically published within an hour.
I'm going ahead and close this issue. If after following the instruction above the library is not published for any reason or if you need further assistance feel free to comment here.

closed 2021-03-03 by cmaglie