arduino/Arduino #11298: [Library Manager] Please add the MyDelay library

mggates39, 2021-02-24:
Please add the MyDelay library. https://github.com/mggates39/MyDelay 
 It is currently release v1.0
This is a mills() based delay library that simplifies using delay timers. it can trigger once, for a specific number of times or forever.  It can also call a callback function for every trigger.

per1234, 2021-02-24:
Hi @mggates39. Thanks for your interest in contributing to the Arduino Library Manager index.

The filename of the library.properties file in your repository is misspelled "ibrary.properties" (note the missing "l"). This will block the addition of your library to the index.

Please fix the filename, then either create a new git tag/GitHub release or else redo the existing tag.

Once that's done, please comment here so we'll know we can proceed.

More information:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

mggates39, 2021-02-24:
The library properties file has been updated.  Release v.1.0.1 has been published.

mggates39, 2021-02-24:
I ran Arduin-lint on the library and fixed the issues found.  Latest release is now v1.02

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie