arduino/Arduino #11300: Lubuntu and STM32 - text messed up in the log window

igor-m, 2021-02-25:
Installed today (Lubuntu 20.04) the latest IDE and the latest STM32 json and I get a mess in the output log text - see below.
Garbage after the "Linking.." line..
It prints fine with Arduino Uno board selected..

[image]

facchinm, 2021-02-26:
Hi @igor-m ,
this happens when the commanline is too long to be correctly displayed. I tried to find a solution but didn't end up with acceptable results (maybe some java guru could help :slightly_smiling_face: ). Apart from that, you can select that line and paste it in another editor/shell and it renders correctly (in case you need to use it).

igor-m, 2021-02-26:
@facchinm: thanks for the info! I've double-checked the length of that line and it is 12300 chars long in the Fade example for the stm32Discovery F407 board :)

igor-m, 2021-02-27:
It seems to me the ie. 12300 chars long line should be written into a separate file..