arduino/Arduino #11303: [Library Manager] Please add AmazingClickButton library

viamgr, 2021-02-26:
Hello,
please add our library to your library manager.

Link to the repository:
https://github.com/viamgr/Arduino-Awesome-Click-Button

And to the library.properties file:
https://github.com/viamgr/Arduino-Awesome-Click-Button/blob/main/library.json

Thanks!

per1234, 2021-02-26:
Hi @viamgr. Thanks for your interest in contributing to the Arduino Library Manager index.

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

viamgr, 2021-02-27:
Thanks. The name of the library changed.
[v1.0.1](https://github.com/viamgr/Awesome-Click-Button/releases/tag/1.0.1)

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie