arduino/Arduino #11304: Suggestion to save the settings in the project

rtek1000, 2021-02-27:
Hello,

I would like to suggest the possibility of saving the settings in the sketch, or in a * .ini or other file, so that it is easier to use the IDE with designs of different board models.

It is quite uncomfortable to have to select the model of the board each time the project is opened, I believe it would also be interesting that the selection of the board is independent of another instance of the IDE.

per1234, 2021-02-27:
Hi @rtek1000 . Thanks for taking the time to submit a this.

We already have a prior issue report about this at https://github.com/arduino/Arduino/issues/1481 (and several others referenced from there).

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.
If you end up with additional information to share, feel free to comment in the other thread.

Note that Arduino CLI, Arduino Pro IDE, and Arduino Web Editor already have support for this via sketch.json (https://arduino.github.io/arduino-cli/latest/sketch-specification/#metadata).

closed 2021-02-27 by per1234

rtek1000, 2021-02-27:
Hello,

Thanks for commenting on the other IDE versions.

I saw the issues open for the Arduino PRO IDE, it still seems a little raw, I hope it will mature soon.

As for the WEB version, sometimes connection instability occurs, so I prefer an offline version.

The CLI version seems to me to go back to the days of DOS, I use shell mainly in linux, but I don't see much advantage in relation to an IDE, anyway I believe it is important for automation through third party IDE.