arduino/Arduino #11305: [Library Manager] Please add the NVSRAM library

gpb01, 2021-02-27:
Please add the NVSRAM library

Repository: https://github.com/gpb01/NVSRAM

This library enables you to read and write into Microchip NVSRAM 23LCV512 (64K x 8 bit) and 23LCV1024 (128K x 8 bit), battery-backed SPI RAM.

Thank you.

per1234, 2021-02-27:
Hi @gpb01. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

gpb01, 2021-02-27:
Thanks, release with tag 1.0.0 added.

cmaglie, 2021-03-03:
Done.

closed 2021-03-03 by cmaglie