arduino/Arduino #11306: Error in compilation by Arduino Mega or Mega 2560.

woodyv, 2021-02-27:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

fork/exec C:\Users\woody\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc.exe: Name or type of file is too long.
Error in compilation by Arduino Mega or Mega 2560.

Please advise me how to change setup to shorten path? I need to compile Marlin firmware for 3D printer.
Thank you Vaclav

matthijskooijman, 2021-02-27:
IIRC there was an issue like this recently fixed, not sure if it was released yet. Can you try the latest released version (1.8.13) and if the issue persists there, maybe also the Hourly Build to see if it is fixed there?

closed 2021-02-27 by woodyv

per1234, 2021-02-27:
Fixed by https://github.com/arduino/arduino-cli/pull/961