arduino/Arduino #11310: [arduino/Arduino] [Library Manager] Change (paragraph) description of Library

mrfaptastic, 2021-02-27:
Hi,

Can you please change the 'paragraph' attribute of an existing library called 'Easy ESP Arduino IoT with CC1101' to:

> An easy to use library to send and receive messages between devices using the Texas Instruments CC1101 Low-Power Sub-1GHz RF Transceiver

Or, refer to: https://github.com/mrfaptastic/Easy-IoT-Arduino-CC1101-LORA/blob/master/library.properties

To refresh library description. Thank you.

per1234, 2021-02-28:
Hi @mrfaptastic. The `paragraph` value is taken directly from the library.properties file of your library. So if you want it changed, you only need to change the value, make sure everything is compliant with the requirements stated in [the Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ), and then make a release/tag.

I see you did change the `paragraph` and made the 1.0.3 release, but what you missed is the compliance part. You changed the `name` value:
https://github.com/mrfaptastic/Easy-IoT-Arduino-CC1101-LORA/commit/b46f35c16d343a0758288f786d70be14b0457ec9#diff-f267f9ae2f5d5d56a1dbb305e95d40d4408d9ee7f71357f1f67fb18c29b7f082
This caused the indexer to reject your 1.0.3 release. Libraries are locked to the `name` value they had when they were added to the Library Manager index. The reason is that the `name` value is the unique identifier for the library.

If you're really set on changing the library name, it is possible to submit a request here for that:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-change-my-librarys-name
Otherwise, just change the name back and then either make another tag/release or redo the existing one.

So please let us know how you would like to proceed.

mrfaptastic, 2021-02-28:
Thank you kindly @per1234 for this really detailed explanation! I will close this issue and raise a name change request.

closed 2021-02-28 by mrfaptastic